beagleboard/pocketbeagle #3: Boot debugging

yuvadm, 2017-09-24:
I'm attempting to boot an existing BeagleBone image I have on an SD card (and then debug to see if there are any require changes for the new board), however I'm not getting any signs of life from the board other than a steady blue LED.

Are there any hardware pins I can use to get serial UART output from the bootloader to debug the boot sequence? Is there any official flow I can follow to get the board to boot from a well-known image?

jadonk, 2017-09-25:
Yes, the UART0 pins are active logging during SPL/u-boot and the same port will also send boot requests (triggered by on-board ROM) if other boot sources are not identified (you can even boot over USB).

To see the console output, I recommend connecting one of these: https://shop.mikroe.com/ftdi-click (available from several distributors).

closed 2017-09-25 by jadonk

yuvadm, 2017-09-25:
@jadonk thanks for the tip! Unfortunately I'm not getting anything on the UART0 pins. I'm connecting the USB connector to a 5V power supply, and I have an FTDI running `screen /dev/ttyUSB0 115200` with RX/TX pairs crossed and a shared ground on any of the GND pins. Nothing is coming up over the wire, neither when I boot from an SD card, nor when there's no SD card at all. Is there anything I'm missing?

Also, can you describe how to boot over USB? That sounds interesting!

Fuegomar, 2017-10-01:
... some tips that might help you

https://debian.beagleboard.org/images/bone-debian-9.1-iot-armhf-2017-09-21-4gb.img.xz 
... working when booting from an unmodified /rootfs/boot/uEnv.txt 

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
#cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh

... then ssh into debian@192.168.7.2 (password=temppwd)
... or use your web browser to 192.138.7.2

However, removing the comment '#' from #cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh so as to read

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh 

and booting from the SDcard will hang all 4 leds and the board gets hot (but does not let out the magic smoke)

// change  /rootfs/etc/ssh/sshd_config for root login as follows...
#LoginGraceTime 2m
#PermitRootLogin prohibit-password // change to PermitRootLogin yes
PermitRootLogin yes
#StrictModes yes
#MaxAuthTries 6
#MaxSessions 10

RobertCNelson, 2017-10-02:
@Fuegomar there is no eMMC on the PocketBeagle, so don't "enable" /opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh  and expect it to do something..

```
However, removing the comment '#' from #cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh so as to read

##enable Generic eMMC Flasher:
##make sure, these tools are installed: dosfstools rsync
cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh
```
Regards,

Fuegomar, 2017-10-02:
Ok and thank you!

wcontello, 2017-10-12:
I am not clear from these posts whether I *must* modify files to get the Pocket Beagle to boot from an SD card flashed with the image "bone-debian-9.2-iot-armhf-2017-10-10-4gb.img.xz" using the tool "Etcher" on mac OS X.  

So the question is... if Etcher completed the flaming procedure onto a 32GB micro SD card and I insure the card in the Pocket Beagle, then power up from a USB port, should it boot?

When I observe is the BLU LED on solid (labeled "PWR")  the four LEDs (labeled "USR23", "USR2", "USR1", and "USR0") never google or flash AND the board gets hot (not too hot to touch, but hot)

RobertCNelson, 2017-10-12:
@wcontello did you press the power button?

wcontello, 2017-10-12:
No.  I did not press any buttons.  I will try that.

Single press, then release... nothing
Press, hold (to count of 1), release, nothing
Press, hold (to count of 2), release, nothing
Press, hold (to count of 5), release, nothing

Any other ideas?

wcontello, 2017-10-12:
If I hold the power button to a count of 10, then the BLU LED turns off.  If I press, then release, then LED turns back on.  So I guess the image is incorrect?

wcontello, 2017-10-12:
I can mount the SD card from a virtual Debian machine on macOS X.  Here is a screenshot of the root folder:
[image]

wcontello, 2017-10-12:
The image shasum matched.

Proton:~ wayne$ shasum -b -a 256 ~/Downloads/bone-debian-9.2-iot-armhf-2017-10-10-4gb.img.xz 
be1eac7a5e526930155520215329a6c39071b82199c0745c300e68b7e6c7180b */Users/wayne/Downloads/bone-debian-9.2-iot-armhf-2017-10-10-4gb.img.xz

RobertCNelson, 2017-10-12:
Humm, that's what i'm getting, why did the script generate a different value..

```
rcnee@debian:~/web/rootfs/bb.org/testing/2017-10-10/stretch-iot$ cat debian-9.2-iot-armhf-2017-10-10.tar.xz.sha256sum
167e8d3d6a4f105c4ae254c0dfb8170caeba9a2a070b2f048b20aa6c8586f584  debian-9.2-iot-armhf-2017-10-10.tar.xz
```

linode:
```
rcnee@debian:~/web/rootfs/bb.org/testing/2017-10-10/stretch-iot$ sha256sum bone-debian-9.2-iot-armhf-2017-10-10-4gb.img.xz
be1eac7a5e526930155520215329a6c39071b82199c0745c300e68b7e6c7180b  bone-debian-9.2-iot-armhf-2017-10-10-4gb.img.xz
```

nas:
```
voodoo@nuc-git:/var/www/html/rcn-ee.us/rootfs/bb.org/testing/2017-10-10/stretch-iot$ sha256sum bone-debian-9.2-iot-armhf-2017-10-10-4gb.img.xz
be1eac7a5e526930155520215329a6c39071b82199c0745c300e68b7e6c7180b  bone-debian-9.2-iot-armhf-2017-10-10-4gb.img.xz
```

Local copy i download that day:
```
voodoo@hades:/opt/images/bb.org/testing/2017-10-10/stretch-iot$ sha256sum bone-debian-9.2-iot-armhf-2017-10-10-4gb.img.xz 
be1eac7a5e526930155520215329a6c39071b82199c0745c300e68b7e6c7180b  bone-debian-9.2-iot-armhf-2017-10-10-4gb.img.xz
```

wcontello, 2017-10-12:
The has matches what's on the website: https://beagleboard.org/latest-images
[image]

wcontello, 2017-10-13:
Should I expect the image "bone-debian-9.2-iot-armhf-2017-10-10-4gb.img.xz" to boot a Pocket Beagle when flashed to an SD card using Etcher?

And what should I expect when connected to a host system running macOS X?

Morantz, 2017-10-13:
I have had no issues booting my PocketBeagle with this image.  I used Etcher on a Win10 Laptop to write image to a class 4 MicroSD card. -  Dan Brown 
"My doctor says that I have a malformed public-duty gland and a natural deficiency in moral fibre, and that I am therefore excused from saving Universes." ~Douglas Adams

   

   

|  | Virus-free. www.avg.com  |

wcontello, 2017-10-14:
I am using a class 10 micro SD card from Lexar

metalheart, 2017-10-14:
I did the same mistake - thought that OSD soc has emmc and uncommented line with emmc flashing - it possibly fried my pocketbeagle. it does not boot anymore and there is no any output on UART0 pins. Did you managed to start board after flashing proper image?

wcontello, 2017-10-14:
I have not modified any files on the image.  Are you saying you think the image is not correct?

jadonk, 2017-10-16:
If the image is correct, this is likely an EEPROM issue. I will get an EEPROM programmer image.

wcontello, 2017-10-16:
I do not understand 

> If the image is correct, this is likely an EEPROM issue. I will get an EEPROM programmer image.

Are you saying you can provide an image that will program an EEPROM on the Pocket Beagle?  Are you saying I should locate, download, and boot to am image that can program the EEPROM?

metalheart, 2017-10-17:
I found WP testpoint on board, does I2C is connected to EEPROM? Is there any information how to recover it? Also board has JTAG testpoints, could it be started under JTAG and access EEPROM IC?

jadonk, 2017-10-18:
Well, the program I use to write the EEPROMs is at https://gist.github.com/jadonk/809337ed8435cdc25f99887746550ed2. There's some instructions there on hacking an image to assume you are booting on a PocketBeagle. Ultimately, I need to break this down into something simpler for you. :-(

wcontello, 2017-10-18:
What if I do not have a BeagleBone Black?

wcontello, 2017-10-18:
Can you suppose how a Pocket Beagle shipped without the EEPROM properly programmed?

wcontello, 2017-10-18:
Can you trade me for the one I have?

yuvadm, 2017-10-19:
@wcontello You are spamming this issue, please take further communications on this topic offline.

ahbushnell, 2017-10-19:
Return it if it's defective.

wcontello, 2017-10-24:
Received another device and it booted perfectly.  I did return the other device

ahbushnell, 2017-10-24:
Great!