beagleboard/pocketbeagle #7: 'FAQ#How_do_I_get_connected_to_the_Internet' missing

easyw, 2017-09-28:
Hi,
in the Faq there is missing the "How_do_I_get_connected_to_the_Internet" part...
https://github.com/beagleboard/pocketbeagle/wiki/FAQ#How_do_I_get_connected_to_the_Internet

Is there any suggestion?

Thx

easyw, 2017-09-29:
What I would need is a PHY to be connected to the pocketbeagle module...

jadonk, 2017-10-02:
See the update on the FAQ page that includes connecting an Ethernet MAC/PHY via SPI. The Ethernet MII pins are not brought out on PocketBeagle and are reserved for you to use in your own re-spin of PocketBeagle.

closed 2017-10-02 by jadonk

easyw, 2017-10-05:
it would be nice if would possible to add this 10/100 BASE-TX ethernet adapter
https://shop.mikroe.com/eth-wiz-click
much better than the eth-click board, which is just a 10BASE-T adapter