beagleboard/pocketbeagle #11: Incorrect pinout diagram label for P2_28 gpio

derekmulcahy, 2017-10-31:
The pinout image for the pocket beagle has the wrong GPIO pin number for P2_28. It is labelled as 124 but the pocket beagle schematic links it to MCASP0_AXR1. This pin is described on P36 of the data sheet as being gpio3_20. The math (3x32+20) would indicate that this is really pin 116. 

The device tree has the correct mapping for P2_28.

`gpio = <0x160 0x14 0x0>;`

The 0x14 is an offset of 20. This mapping works fine with config-pin to toggle the P2_28 gpio.

derekmulcahy, 2017-11-01:
I also think the label for P1_33 should be GPIO 111 and not 110.

derekmulcahy, 2017-11-01:
Updated image attached with P1_33 = GPIO 111, P2_24 = GPIO 44, P2_28 = GPIO 116

[image]

jadonk, 2017-11-02:
P1_33 is labelled as as "1" under "PRU0" on the silkscreen. GPIO3.15 == GPIO111, so yes, it is GPIO111. Are you referencing something in the schematic? @derekmulcahy, which document is wrong??

jadonk, 2017-11-02:
Ah, that diagram. Thanks!

derekmulcahy, 2017-11-02:
Sorry, I didn't say it was on the Wiki FAQ page.

jadonk, 2017-11-02:
Did I get it all? https://github.com/beagleboard/pocketbeagle/commit/cc3af73758f25dcc473f226d1892a21de610f96f

derekmulcahy, 2017-11-02:
Yes, looks good.

jadonk, 2017-11-02:
OK, I should have it updated on the wiki now too.

closed 2017-11-02 by jadonk