beagleboard/pocketbeagle #12: Wrong defaults in pinout diagram?

shirriff, 2017-11-15:
In the pinout diagram (PocketBeagle_pinout.png), do the dotted lines indicate the default mode? If so, the PRU defaults are wrong (P1_29/31/33/35 and P2_28/30/32/34).

I checked the latest Stretch IoT build (Linux beaglebone 4.4.91-ti-r133 #1 SMP Tue Oct 10 05:18:08 UTC 2017 armv7l GNU/Linux) and these pins come up in GPIO mode, not PRU.

jadonk, 2017-12-06:
@RobertCNelson can you make sure the default modes match the silkscreen in all of the images? The device tree I provided made "default" be the mode that is documented as the default. This will help with many newb issues.

RobertCNelson, 2017-12-07:
@shirriff and @jadonk correct pinmux's should now be avaialbe in:

```
4.4.91-ti-r138 (building)
4.9.67-ti-r82 (building)
4.14.4-ti-r15 (building)
4.14.4-bone7 (building)
&
4.15.0-rc3 (sunday)
```
and the matching config-pin will be:
```
bb-cape-overlays_4.4.20171207.0-0rcnee1
```
https://github.com/beagleboard/bb.org-overlays/commit/0013331fd0fe49b90af5a57944ef854283ab1d70

Regards,

closed 2018-01-01 by jadonk