beagleboard/pocketbeagle #13: GPIO 44 labelled as 48 in silkscreen

brucepom, 2017-11-28:
Thanks for an amazing board!

In my testing the pin labelled as 48 in the silkscreen is actually accessed as pin 44.

[image]

I haven't found an official pinout illustration for PocketBeagle but I did find this one: http://files.linuxgizmos.com/bboard_pocketbeagle_pinout.jpg I'm unsure of the original source of this image but the image also describes that pin as 44.

RobertCNelson, 2017-11-28:
@brucepom correct it's pin 44:

https://github.com/beagleboard/pocketbeagle/issues/4

https://github.com/beagleboard/pocketbeagle/wiki/FAQ#What_is_the_pinout_for_the_expansion_headers

Regards,

mwelling, 2017-11-28:
This has already been corrected on the design files. Guessing that when the next production run commences they will update them. Might be a while though.

https://github.com/beagleboard/pocketbeagle/pull/8

brucepom, 2017-11-29:
Thanks guys, looks like you were already on top of it. Really appreciate all your work.

closed 2017-12-06 by jadonk