beagleboard/pocketbeagle #14: PinOut Names not working corectly

el-j, 2017-12-08:
first of all: really nice board! thank you!

i have a problem with bonescript and the pin names. i try to use the gpio pins from p2. 
after a frustrating time where nothing worked with the board. i figured out that i have to update the bonescript module with the latest release from github. (it was not integrated in the lastest debian image)

now i am able to use the "P1_xx" "P2_xx" declaration for the pins without getting just an error. 
sadly it does not work properly for all pins, from my testing. 

the pin P2_2 and P2_8 working correctly with a pushbutton. 
but the pins P2_4 + 6 + 10 do not get any input with the same code. 

i try to find the answer with the getPinMode methode but i cannot get a clue out of it.

not working:
```
{ pin: 'P2_6',
  name: 'GPIO1_25',
  options:
   [ 'gpmc_wpn',
     'gmii2_rxer',
     'gpmc_csn5',
     'rmii2_rxer',
     'mmc2_sdcd',
     'pr1_mii1_txen',
     'uart4_txd',
     'gpio1_25' ],
  gpio: { active: true, direction: 'in', allocated: false } }
mux = undefined
pullup = undefined
slew = undefined
gpio = [object Object]
options = gpmc_wpn,gmii2_rxer,gpmc_csn5,rmii2_rxer,mmc2_sdcd,pr1_mii1_txen,uart4_txd,gpio1_25
pin = P2_6
name = GPIO1_25
err = undefined
```
working:
```
{ pin: 'P2_8',
  name: 'GPIO1_28',
  options:
   [ 'gpmc_ben1',
     'mii2_col',
     'NA',
     'mmc2_dat3',
     'NA',
     'NA',
     'mcasp0_aclkr',
     'gpio1_28' ],
  gpio: { active: true, direction: 'in', allocated: false } }
mux = undefined
pullup = undefined
slew = undefined
gpio = [object Object]
options = gpmc_ben1,mii2_col,NA,mmc2_dat3,NA,NA,mcasp0_aclkr,gpio1_28
pin = P2_8
name = GPIO1_28
err = undefined
```
do you have any idea what i can do, to use this pins?

jadonk, 2017-12-19:
```sh
$ sudo npm install -g --unsafe-perm bonescript@beta
[sudo] password for debian:
npm WARN deprecated coffee-script@1.9.1: CoffeeScript on NPM has moved to "coffeescript" (no hyphen)

> ref@1.3.5 install /usr/local/lib/node_modules/bonescript/node_modules/ref
> node-gyp rebuild

make: Entering directory '/usr/local/lib/node_modules/bonescript/node_modules/ref/build'
  CXX(target) Release/obj.target/binding/src/binding.o
  SOLINK_MODULE(target) Release/obj.target/binding.node
  COPY Release/binding.node
make: Leaving directory '/usr/local/lib/node_modules/bonescript/node_modules/ref/build'
- graceful-readlink@1.0.1 node_modules/bonescript/node_modules/graceful-readlink
/usr/local/lib
└─┬ bonescript@0.6.2-beta2
  ├─┬ express@4.13.4
  │ ├─┬ accepts@1.2.13
  │ │ └─┬ mime-types@2.1.17
  │ │   └── mime-db@1.30.0
  │ ├── content-type@1.0.4
  │ ├── depd@1.1.1
  │ ├── parseurl@1.3.2
  │ └─┬ proxy-addr@1.0.10
  │   └── forwarded@0.1.2
  ├─┬ ffi@2.0.0
  │ └── ref@1.3.5
  ├─┬ serialport@3.1.2
  │ ├── commander@2.12.2
  │ └─┬ object.assign@4.0.4
  │   └── function-bind@1.1.1
  └─┬ winston@2.1.1
    └── stack-trace@0.0.10

$ DEBUG=true node -pe "require('bonescript').getPinMode('P2_6')"              
4:4:91-ti-r133
Using Mainline interface
debug: index.js loaded
debug: getPinMode(P2_6);
debug: getPinMode(P2_6): Error: EACCES: permission denied, open '/sys/kernel/debug/pinctrl/44e10800.pinmux/pins'
{ pin: 'P2_6',
  name: 'GPIO1_25',
  options:
   [ 'gpmc_wpn',
     'gmii2_rxer',
     'gpmc_csn5',
     'rmii2_rxer',
     'mmc2_sdcd',
     'pr1_mii1_txen',
     'uart4_txd',
     'gpio1_25' ],
  gpio: { allocated: false } }
debug: Calling run()

Exiting Cleanly
Unable to open /dev/mem
Unable to open /dev/mem
ERROR: trying to disable motors before they have been initialized
Unable to open /dev/mem
Unable to open /dev/mem
Unable to open /dev/mem
$ DEBUG=true sudo node -pe "require('bonescript').getPinMode('P2_6')"         
{ pin: 'P2_6',
  name: 'GPIO1_25',
  options:
   [ 'gpmc_wpn',
     'gmii2_rxer',
     'gpmc_csn5',
     'rmii2_rxer',
     'mmc2_sdcd',
     'pr1_mii1_txen',
     'uart4_txd',
     'gpio1_25' ],
  gpio: { allocated: false },
  mux: 7,
  slew: 'fast',
  rx: 'enabled',
  pullup: 'pullup' }

Exiting Cleanly
ERROR: trying to disable motors before they have been initialized
$ DEBUG=true node -pe "require('bonescript').getPinMode('P2_8')"
4:4:91-ti-r133
Using Mainline interface
debug: index.js loaded
debug: getPinMode(P2_8);
debug: getPinMode(P2_8): Error: EACCES: permission denied, open '/sys/kernel/debug/pinctrl/44e10800.pinmux/pins'
{ pin: 'P2_8',
  name: 'GPIO1_28',
  options:
   [ 'gpmc_ben1',
     'mii2_col',
     'NA',
     'mmc2_dat3',
     'NA',
     'NA',
     'mcasp0_aclkr',
     'gpio1_28' ],
  gpio: { allocated: false } }
debug: Calling run()

Exiting Cleanly
Unable to open /dev/mem
Unable to open /dev/mem
ERROR: trying to disable motors before they have been initialized
Unable to open /dev/mem
Unable to open /dev/mem
Unable to open /dev/mem
```

So, it seems P2_8 and P2_6 have the same issue, which is the pinmux setting cannot be read without being superuser.

jadonk, 2017-12-19:
Possible fix:

```
# /etc/udev/rules.d/83-pinmux-noroot.rules
#
# Change group to gpio
KERNEL=="44e10800.pinmux", PROGRAM="/bin/sh -c '/bin/chown -R root:gpio /sys/kernel/debug/pinctrl/44e10800.pinmux'"
# Change user permissions to ensure user and group have read/write permissions
KERNEL=="44e10800.pinmux", PROGRAM="/bin/sh -c '/bin/chmod ugo+x /sys/kernel/debug; /bin/chmod -R ug+rw /sys/kernel/debug/pinctrl/44e10800.pinmux'"
```

jadonk, 2017-12-19:
This is really a BoneScript (and Debian image) issue, not PocketBeagle hardware, so I've opened an issue on BoneScript and will close this one.

closed 2017-12-19 by jadonk

RobertCNelson, 2017-12-20:
Doh! Bonescript goes thru the /sys/kernel/debug/ interface!  i just had gpio user setup on /sys/class/gpio.. okay i'll patch bb-customization with the new udev rule...