beagleboard/pocketbeagle #16: C6 on silkscreen is C7 on schematic

sirkha, 2018-01-10:
I thought that C7 seemed awfully far away from P1_17 and P1_18, so I checked. On the [SiP B OSD3358 SiP JTAG, USB & Analog Signals](https://github.com/beagleboard/pocketbeagle/wiki/images/SiP-B-sch.png), C7 probes out to where [C6 is on the silkscreen, right next to the SIP](https://github.com/beagleboard/pocketbeagle/wiki/images/21fig-PB-walkaround.png).

[image]
[image]

jadonk, 2018-01-23:
I confirmed the existing silkscreen is correct with regards to C6 and C7. Both capacitors have GND on one side and a supply rail on the other (for decoupling). Perhaps you experienced some bleed-over between the supply rails? [Careful with this, you might damage the device.] You might try measuring the voltages instead of looking at continuity. You should C6 have 3.3V across it and C7 have 1.8V across it.

Also, you can tell visually that C6 is the larger capacity simply by the size, so that should be a good indicator for you. I looked at the traces on the GERBERs and they go to the proper places without shorts, so I'm confident there is no issue here.

closed 2018-01-23 by jadonk

mwelling, 2018-01-24:
Besides the sub-optimal placement of C7. :)