beagleboard/pocketbeagle #19: Can't boot from SD card

iSsO, 2018-04-18:
I think my board is booting from the wrong mmc

I've downloaded the latest image from http://beagleboard.org/latest-images for pocketbeagle named Debian 9.3 2018-03-05 4GB SD IoT, then flashed the image to an 8gb sd card using etcher. The board refuses to boot after inserting the card and powering it.

Only the blue PWR led is on, and the octavo chip gets hot.

When I connected a serial monitor to UART0, I found that it is trying to boot from MMC1 not MMC0 (the SD card)
[image]

jadonk, 2018-05-20:
@beagleboard/kernel-contributors Is this a case of a bad board or, more likely in my view, a bad SD card?

@iSsO can you try a different SD card?

iSsO, 2018-06-14:
I've tried different images and SD cards with the same result.

Could it be a problem with the EEPROM?

jadonk, 2018-06-22:
Do you have a Linux host? Can you try: https://elinux.org/images/b/b2/Netconsole.pdf http://bbb.io/pbncgist ?

jadonk, 2018-07-10:
Unclear if it can be fixed. Depends if the problem is in the SIP or on the board. We have a way to boot it via USB using node-beagle-boot.

https://schd.ws/hosted_files/elciotna18/42/netconsole.pdf
https://www.youtube.com/watch?v=9CxfKfjykz4

You can then login and do various tests on the MMC interface. You could probably change the device tree to make it SPI-mode.

Anyway, the above is mostly FYI on how we'd do any failure analysis. Maybe best to contact me via my e-mail address listed at bbb.io/about and we'll work with distributor and manufacturer to get you a replacement.

iSsO, 2018-07-15:
Thanks Jason, thanks for the NetConsole stuff, I didn't know that existed!

I'll close the issue for now, will contact you if I need additional assistance.

closed 2018-07-15 by iSsO