beagleboard/pocketbeagle #21: issue about the CAN Bus of Pocketbeagle

zerulin, 2018-07-10:
Please can anybody help me? I try to use the can bus with SN65HVD230 can module. I enable the can pins by:
config-pin P2_09 can
config-pin P2_11 can
And then I test loopback, it works. But if I try to communicate with other CAN module, there are no data transmitted or received when I check with ifconfig can0. ( I also try with other CAN pins, but it still cannot work). Do I need other configuration or what' wrong with my operation?

zerulin, 2018-07-10:
BTW, I use the newest Debian 9.4

zerulin, 2018-07-10:
[image]
This is one of my connection, the module is https://www.waveshare.com/sn65hvd230-can-board.htm
I will test again and attach the images of commands

zerulin, 2018-07-10:
[image]
This is loopback test, it works.
But if I off the loopback mode, and try to send data, there is no data send via Tx when I check ifconfig can0:
[image]

zerulin, 2018-07-10:
[image]
Debian version I use

mwelling, 2018-07-10:
Are you connecting the CAN transceiver to another CAN device?

zerulin, 2018-07-10:
:( ， I connect to another BB Black with CAN CAPE. I have test the BB Black +CAPE x2, they are work, and if I change to the Pocketbeagle, It failed.

jadonk, 2019-01-09:
Can you try a different PHY?

https://spin.atomicobject.com/2017/05/06/troubleshooting-can-bus-communication-subaru/

Jabed-Akhtar, 2021-11-24:
In my case, I am not using CAN-transceiver at all, and I am able to monitor CAN-Messages over the wire. But I am not able to send out CAN-Messages. Could it be because of not using CAN-Transceiver? Can anyone please provide me any Datasheet or any Document, where I can see how the CAN is built in PocketBeagle!
Some more questions:
> Is CAN-Transceiver build-in inside the PocketBegle?
> Is the 120Ohm for CAN also build-in inside the PocketBegle?

Tanmay-FuelDaddy, 2023-05-23:
I too would appreciate an answer to  jabed's Questions

gletchm, 2023-05-23:
I am pretty sure you need the transceiver. The CAN bus in a 2 wire differential bus. In my experience that CAN peripheral in the chip, expects to connected to a CAN transceiver. I can see how receiving CAN message could work, sending is a lot different. You have to control the signaling correctly. They make CAN transceiver that will work with 3.3V, if that is the issue. Here is a link to some good info: https://www.beyondlogic.org/adding-can-to-the-beaglebone-black/

Jabed-Akhtar, 2023-06-07:
> I am pretty sure you need the transceiver. The CAN bus in a 2 wire differential bus. In my experience that CAN peripheral in the chip, expects to connected to a CAN transceiver. I can see how [...]

Thank you so much :)

RobertPHeller, 2023-06-25:
Have a look at https://github.com/RobertPHeller/LCC-PNET-Router for a board that connects to both CAN IFs on a PocketBeagle. One CAN IF (can0) connects to LCC's CAN Network and the other CAN IF (can1) connects to Pricom PNET.  This board works properly (I have some boards make and have tested it).