beagleboard/pocketbeagle #24: Pocketbeagle not booting from SD card

MariusTOR, 2018-12-12:
I'm trying to evaluate a PocketBeagle (rev A2) for use in an instrumentation class I'm teaching this Spring, however, the one I have refuses to boot.  I just purchased it from Arrow, and the microSD card that I'm using to boot it has never had anything else written to it (save by the manufacturer).

Using the Etcher program suggested by the quick start guide, I've burned a 16GB microSD card once with the image for Debian Linux v9.5 and, after that failed, with Debian Linux v9.1.  Both times, the only change I've done is to edit /boot/uEnv.txt as directed to make sure that cmdline=init=/opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh is uncommented.  

Booting with Debian v9.1
Upon plugging in the PB to USB, all 4 LEDs light up momentarily.  They all go off, and USR0 starts its heartbeat pattern.  USR1 starts to blink intermittantly and the Octavo chip starts to heat up.  After about 5 seconds, all four LEDs light up simultaneously.  I let it stay that way for 5-10 more seconds before the Octavo chip gets too warm and unplug the USB.

Booting with Debian v9.5
1 - Upon plugging in the PB to USB, all 4 LEDs light up momentarily.  
2 - They all go off, and USR0 starts its heartbeat pattern and USR2 is on.  USR1 starts to blink intermittantly.  
3 - After about 20 seconds, USR2 blinks quickly and intermittently while being dim, USR1 stays off, and USR0 continues its heartbeat pattern.
4 - The Octavo chip is starting to get warm.  I let it stay that way for 30 more seconds and unplug the USB.

In neither case does the board get recognized from a computer running Ubuntu 16.10.  Attempts to browse to the board by going to 192.168.6.2 (while booting with Debian v9.5) failed, but resulted in all 4 LEDs lighting up simultaneously after step 2 of the aforementioned boot process.

I'm at quite a loss as to how to proceed.  Any help would be appreciated.

mwelling, 2018-12-12:
> I'm trying to evaluate a PocketBeagle (rev A2) for use in an
> instrumentation class I'm teaching this Spring, however, the one I have
> refuses to boot. I just purchased it from Arrow, and the microSD [...]

There is no eMMC on the PocketBeagle so this line should remain commented.

> Booting with Debian v9.1
> Upon plugging in the PB to USB, all 4 LEDs light up momentarily. They all
> go off, and USR0 starts its heartbeat pattern. USR1 starts to blink
> intermittantly and the Octavo [...]

Just recomment the line mention above and see if that helps.

> <https://github.com/beagleboard/pocketbeagle/issues/24>, or mute the
> thread

MariusTOR, 2018-12-12:
No good -- I recommented that line and bootup behavior is exactly the same as before.

mwelling, 2018-12-13:
Try reflashing the image to the MicroSD from scratch and plugging it in without modification.

MariusTOR, 2018-12-13:
Found the problem -- I was using a USB cord that only had wires for power, not data.  Thanks!

mwelling, 2018-12-13:
No problem.