beagleboard/pocketbeagle #25: Pocketbeagle not booting from SD card

nutilius, 2019-01-16:
Hello everybody.
Yesterday I receiver PockedBeagle from Farnell and try to boot from SD card. Only Power LED is on, the rest is dead. I have tried 5 different cards with different size and 3 diffrent USB cables. 
System is dead. Whe system has card included, on attached linux system there is no USB device (no messages in dmesg also). I've tried two diffrent official Debian images from - current and last previous.
**When  I remove the card** and again attach board, I see new device  named Texas Instruments and dmesg also presents messages abut ne device.

jadonk, 2019-01-16:
@nutilius The boot image is bad. The device has a USB boot mode in ROM, which is why you'd see it on your computer even without a valid boot image. You can possibly use BeagleBoot.

What steps are you following to program the microSD card?

nutilius, 2019-01-16:
@jadonk Downloaded image was written to SD card by dd tool and later sync command was executed.

nutilius, 2019-01-16:
Of course after decompressing using xz and veryfing checksum.

nutilius, 2019-02-27:
The new part works good. Probably previous was damaged.

closed 2019-02-27 by nutilius