beagleboard/pocketbeagle #29: Question on Beaglebone pocket project

kissup77, 2020-07-02:
We tried to duplicate your pocket Hardware.
Our produced board:
When power is connected, only one LED comes up
Original Pocket Beagle:
When power is connected, only one LED comes up and then other 4 LEDs are up sequencially.

Should we fuse ROM? or EEPROM? to boot it up?
and how to?
Can you share the link if there is any guide for it?

mwelling, 2020-07-02:
You will need the "flasher" image to program the EEPROM and a shunt to enable writing.

https://beagleboard.org/latest-images

Check issue resolved issue #3 here: https://github.com/beagleboard/pocketbeagle/issues/3

mwelling, 2020-07-02:
Another route would be to modify the bootloader such that it doesn't rely on the EEPROM to boot.

kissup77, 2020-07-02:
Can I know what Another route is ? (Physical connection or jumper ?)

mwelling, 2020-07-02:
It is a purely firmware route. The standard Beagleboard U-Boot uses the EEPROM to determine the board type. You can disable this detection and boot whatever you want. This off course requires some firmware chops and understanding of U-Boot.

mwelling, 2020-07-02:
The "flasher" image will boot the board without the EEPROM program as it has a hack in the bootloader already to fall back to defaults for booting the pocketbeagle.

kissup77, 2020-07-03:
I had tried original flash image fused tf card chip, still not booting up. So for beagle newbie, it is recommendable to fuse eeprom?

kissup77, 2020-07-03:
Please tell me how to flash to EEPROM.

mwelling, 2020-07-03:
You need to boot the flasher image in order to program the EEPROM.

If you have access to the debug UART check if you have output there. If not you might have deeper issues.

kissup77, 2020-07-06:
`board_name=[A335BLNK] ...
30 bytes read in 29 ms (1000 Bytes/s)
Loaded environment from /boot/.eeprom.txt
Setting bus to 0
0000: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
0010: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
Setting bus to 0
Setting bus to 0
Setting bus to 0
0000: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
0010: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff    ................
resetting ...

U-Boot SPL 2019.04-00003-g2bf638a03e (Mar 06 2020 - 11:25:08 -0600)
Trying to boot from MMC1
Loading Environment from EXT4... ** File not found /boot/uboot.env **

** Unable to read "/boot/uboot.env" from mmc0:1 **

U-Boot 2019.04-00003-g2bf638a03e (Mar 06 2020 - 11:25:08 -0600), Build: jenkins-github_Bootloader-Builder-137

CPU  : AM335X-GP rev 2.1
I2C:   ready
DRAM:  512 MiB
No match for driver 'omap_hsmmc'
No match for driver 'omap_hsmmc'
Some drivers were not found
Reset Source: Global warm SW reset has occurred.
Reset Source: Power-on reset has occurred.
RTC 32KCLK Source: External.
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Loading Environment from EXT4... ** File not found /boot/uboot.env **

** Unable to read "/boot/uboot.env" from mmc0:1 **
<ethaddr> not set. Validating first E-fuse MAC
BeagleBone: cape eeprom: i2c_probe: 0x54:
BeagleBone: cape eeprom: i2c_probe: 0x55:
BeagleBone: cape eeprom: i2c_probe: 0x56:
BeagleBone: cape eeprom: i2c_probe: 0x57:
Net:   usb_ether
Press SPACE to abort autoboot in 2 seconds`

I am getting this error.

kissup77, 2020-07-06:
Another question from earlier answer:
It is a purely firmware route. The standard Beagleboard U-Boot uses the EEPROM to determine the board type. You can disable this detection and boot whatever you want. This off course requires some firmware chops and understanding of U-Boot.
==> 1. Can you share the link how to disable detection?
==> 2. How to boot with SD card?

mwelling, 2020-07-07:
1) There are some pointers here:
https://octavosystems.com/app_notes/osd335x-eeprom-during-boot/

2) You are booting SD if you have U-Boot coming up.
Remember you have to GND the WP pin which writing to EEPROM.

kissup77, 2020-07-08:
Thank you.
Solved.

kissup77, 2020-07-09:
I can't find the WP pin of the Beagle Board Black wireless and Beagle Board AI.
Is there anything I have to do to write the EEPROM?

mwelling, 2020-07-09:
The BBBW has an external EEPROM and WP test pad for pulling it to ground.
[image]

The BBAI is special creature using the newer AM5728/9. I don't think has an EEPROM.

kissup77, 2020-07-13:
I had duplicated beagle pocket board and 1 board is working fine with your suggestion(WP to GND) while other 4 boards does not come up.
in pocket beagle EMU0 (MISC EMULATION PIN) is not going up to 3.3V while normal board goes up to 3.3V.
What might cause this issue?
Any suggestion?