beagleboard/pocketbeagle #31: No drive

braunmarco, 2020-09-13:
Hello,

i have a pocketbeagle rev. 18355 i installed the image: https://debian.beagleboard.org/images/bone-debian-10.3-iot-armhf-2020-04-06-4gb.img.xz. The beagle is booting but no drive is connected to my windows 10 system. I tried on two systems but it doesnt work. 

I dont know what's the problem? someone can help?

thanks

seamusdemora, 2020-09-16:
I use Linux & Mac, BUT: I don't see a "drive" either... what makes you think you're supposed to see a drive?

johnbacas, 2020-12-05:
I have the same problem.
[image]
In the instructions in the "System Reference Manual" it clearly says we should expect to see a drive...