beagleboard/pocketbeagle #32: Graphics out

SinanAkkoyun, 2022-11-21:
Hello!
I need to add an HDMI output to the pocketbeagle. (even if it requires a chip like the NXP TDA)

How can I go about that and how could I make it work with linux?
Thank you so much!

mwelling, 2022-11-21:
Are you designing a new board based on the SiP or just trying to design an adapter for the PocketBeagle?

PocketBeagle does not break out all of the SiP's LCD display lines to headers you would need a display adapter from the available peripherals. USB might be able to do it but it won't be particularly performant.

If you are making a new board based on the SiP you can leverage the design from other Beagles (if you can source the chips):
[image]

SinanAkkoyun, 2023-01-04:
I want to design a new board based on the SiP and reflow the chip from a pocket beagle

Thank you so much for the provided schematic!

closed 2023-09-19 by jadonk